ColinWaddell/FlightTracker #4: Trying to run based on readme instructions...

tjhall9, 2022-01-04:
When running with `sudo python its-a-plane.py`:

> 
Traceback (most recent call last):
  File "its-a-plane.py", line 1, in <module>
    from display import Display
  File "/home/pi/its-a-plane-python/display/__init__.py", line 4, in <module>
    from utilities.animator import Animator
  File "/home/pi/its-a-plane-python/utilities/animator.py", line 78
    print(f"method1 {frame}")
                           ^
SyntaxError: invalid syntax

When running with `sudo python3 its-a-plane.py`:
Traceback (most recent call last):
  File "its-a-plane.py", line 1, in <module>
    from display import Display
  File "/home/pi/its-a-plane-python/display/__init__.py", line 5, in <module>
    from utilities.overhead import Overhead
  File "/home/pi/its-a-plane-python/utilities/overhead.py", line 1, in <module>
    from FlightRadar24.api import FlightRadar24API
ModuleNotFoundError: No module named 'FlightRadar24'

This is the most recent version of code as of 1/4/22, only APIs downloaded is the one listed in the readme

ColinWaddell, 2022-01-05:
My guess is you've got the flight radar API installed for python 2, not
python 3

ColinWaddell, 2022-01-05:
Yeah. Just read the instructions and they need tweaked to make it more
obvious to use python3

Try:

sudo pip3 install FlightRadarAPI
sudo python3 its-a-plane.py

tjhall9, 2022-01-05:
Finally working, thanks a bunch :)

closed 2022-01-05 by tjhall9

ColinWaddell, 2022-01-06:
No worries @tjhall9 

I've updated the instructions based on your comments. Cheers for your input 🏆 

https://github.com/ColinWaddell/its-a-plane-python/commit/c8f991e37eeece45ab03b3b03cf16e1b76fde87f#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5L11