ColinWaddell/FlightTracker #6: sudo python3 its-a-plane.py does not show data on matrix

graememilne, 2022-04-17:
sudo python3 its-a-plane.py,  if i type this, the matrix does not show anything. At the bottom of the command ,to exit ctrl +c shows. Which to me it means no errors and it is running ? ( cd its-a-plane-python then type sudo python3 its-a-plane.py)
I have set it up to start at bootup and checked it with no errors showing.
The funny thing is, i use the demo command it works and it shows the data of aircraft etc on the matrix .
(cd rpi-rgb-led-matrix ,cd examples-api-use then type as below)
sudo ./demo -D1 --led-rows=32 --led-cols=64
I wonder if you can give me any hints to why the matrix does not show data using
sudo python3 its-a-plane.py
plane.log file is empty.  

I have set it up in “convenience” setting

[Aberdeen config.txt](https://github.com/ColinWaddell/its-a-plane-python/files/8502174/Aberdeen.config.txt)

jgoldburn, 2022-04-25:
I had this exact issue the other day and I debated about whether to post anything, for fear I was being silly and that it was something everyone would know. So I'm glad you've posted this!

For me, the issue related to the hardware that is written in the code. Open the following file:

/display/__init__.py

Line 52 -  options.hardware_mapping = "adafruit-hat-pwm"

You need to change this last part to simply "adafruit-hat"

Save the .py and reboot. This worked for me!

graememilne, 2022-04-25:
Thank you very much jgoldburn for your reply .
This is so much appreciated .
You have solved the problem ,it is now working the way it should be doing now.
I can not thank you enough for this .
Kind Regards
Graeme Milne

ColinWaddell, 2022-04-26:
I've updated the code and added a config option so you can enable/disable the HAT PWM functionality.

In the config you can say `HAT_PWM_ENABLED = True` or `HAT_PWM_ENABLED = False`

The default assumption is you've added the solder bridge.

closed 2022-04-26 by ColinWaddell