ColinWaddell/FlightTracker #7: Error Trying To Start Program

kuffcakes, 2022-12-01:
Hi,

Thanks for the response on Instagram.
Here is the error code I receive when I try and run its-a-plane.py

pi@raspberrypi:~/its-a-plane-python $ sudo python3 its-a-plane.py
Traceback (most recent call last):
File "/home/pi/its-a-plane-python/its-a-plane.py", line 1, in <module>
from display import Display
File "/home/pi/its-a-plane-python/display/__init__.py", line 5, in <module>
from utilities.overhead import Overhead
File "/home/pi/its-a-plane-python/utilities/overhead.py", line 8, in <module>
from config import MIN_ALTITUDE
File "/home/pi/its-a-plane-python/config.py", line 3
"tl_x": -112.351105 # Top-Left Longitude (deg)

SyntaxError: invalid syntax

Here are the contents of my config file:

ZONE_HOME = {
"tl_y": 34.820843 # Top-Left Latitude (deg)
"tl_x": -112.351105 # Top-Left Longitude (deg)
"br_y": 34.381565 # Bottom-Right Latitude (deg)
"br_x": -112.319651 # Bottom-Right Longitude (deg)
}
LOCATION_HOME = [
34.616279 # Latitude (deg)
-112.349679 # Longitude (deg)
0.01781 # Altitude
]
TEMPERATURE_LOCATION = "Prescott Valley"
OPENWEATHER_API_KEY = "836ded039f5960c548b45db693504630"
TEMPERATURE_UNITS = "imperial"
MIN_ALTITUDE = 100
BRIGHTNESS = 50
GPI_SLOWDOWN = 2
JOURNEY_CODE_SELECTED = "PRC"
JOURNEY_BLANK_FILLER = " ? "
HAT_PWM_ENABLED = True

ColinWaddell, 2022-12-01:
At first glance, it looks like you're missing commas after every entry in
each of your dictionaries. Have a look at the example config and note how
dictionaries are formatted

kuffcakes, 2022-12-01:
I can't believe I missed those! I also wrote GPI_SLOWDOWN instead of GPIO. It is working now!

Thank you.

ColinWaddell, 2022-12-01:
No worries, glad you got it working!

closed 2022-12-01 by ColinWaddell