ColinWaddell/FlightTracker #9: Indentation error and no display

Rudy-2707, 2024-01-05:
Hi Colin, thanks for the response on LinkedIn

I had used the same instructions given to run the source code, so cloning the repo, installing the FR24 API dependency and added a config.py file

Prior to this, I had returned to a "blank slate" as you mentioned, running through the installation from process. Not changing any of the files, I ran the code again and this time did not receive any indentation error, but nothing displayed on the LED board. Debugging to check if the code was passing through again, I was able to see that flight information was passing through the terminal as I showed before. 

I will of course re-try to run through the installation process once again and see if there are any errors popping up or the display not working as intended. Will make sure to do this asap as show the errors or if the display doesn't work.

Anyhow, before running through the installation process, I tried using the official demos provided by Adafruit to see if the LED board or RGB bonnet had any issues. I was able to get all the pre-written demos working, as shown below. 

https://github.com/ColinWaddell/its-a-plane-python/assets/74123718/ea7a80d8-0afe-437c-bf84-4ad4ca25a8d0

ColinWaddell, 2024-01-05:
Can you also pop your config.py file in here? Feel free to jiggle the
numbers around a little if you don’t want to pass along your location, etc

When you run the it’s-a-plane.py script does it say “press ctrl-c to quit”?

Rudy-2707, 2024-01-06:
Sure, 

ZONE_HOME = {
    "tl_y": 1.38402, # Top-Left Latitude (deg)
    "tl_x": 103.99256, # Top-Left Longitude (deg)
    "br_y": 1.32872, # Bottom-Right Latitude (deg)
    "br_x": 103.98498 # Bottom-Right Longitude (deg)
}
LOCATION_HOME = [
    1.3545, # Latitude (deg)
    103.9886, # Longitude (deg)
    3.048 # Altitude (km)
]
TEMPERATURE_LOCATION = "Singapore"
OPENWEATHER_API_KEY = "0ad87fe766e217fe21ff31bd9a00bef7" # Get an API key from >
TEMPERATURE_UNITS = "metric"
MIN_ALTITUDE = 100
BRIGHTNESS = 50
GPIO_SLOWDOWN = 2
JOURNEY_CODE_SELECTED = "SIN"
JOURNEY_BLANK_FILLER = " ? "
HAT_PWM_ENABLED = True

After running its-a-plane.py it says "press ctrl-c to quit" but nothing shows up on the LED board, like in the video below. 

https://github.com/ColinWaddell/its-a-plane-python/assets/74123718/0f7c0d5d-71ed-49f2-a186-7d5043f4b9cd

ColinWaddell, 2024-01-06:
I see you have HAT_PWM_ENABLED = True in the config. Have you performed the
soldering required first before using this settings?

https://learn.adafruit.com/assets/57727

Rudy-2707, 2024-01-06:
Ahh, small detail that I completely missed out on. Didn't need it so I didn't bother soldering. Program finally loaded with the correct flight info 

Thank you!!

closed 2024-01-06 by Rudy-2707

ColinWaddell, 2024-01-06:
Amazing! Excellent work persevering with this, glad we got it resolved
quickly. I’ll add a little “troubleshooting” section to the README as
you’re not the first to be caught out by this.