ColinWaddell/FlightTracker #12: Hardwiring Help Needed

HighProofTechno, 2024-08-24:
* I assume that when installing the 330 ohm resistor, the trace on the underside of the veroboard needs to be cut, otherwise the resistor is just shorted
* Are the two black 18 gauge wires on the veroboard tied together on the underside of the board?

ColinWaddell, 2024-08-25:
Both assumptions are correct. All wires of the same colour have continuity
and the track under the resistor is split

If you bought a starter-soldering pack they often come with 5mm drill bit
on a handle, that’s the best thing for splitting tracks on the
prototyping-board. Just give it a couple of turns where you need to cut a
track and it’ll go through with no bother.

ColinWaddell, 2024-08-25:
…..go through the copper with no bother. No need to go all the way through
the FR4

HighProofTechno, 2024-08-25:
Colin;
Many thanks for the response.  I'm a novice at Raspberry Pi projects and I'm learning a lot.
I appreciate your patience on this.
 
 
Regards,
Tom