ColinWaddell/FlightTracker #14: its-a-plane.py won´t start

cooool39, 2025-07-04:
Hi, i can´t start the "its-a-plane.py"

`
pi@raspberrypi:~/its-a-plane-python $ sudo python3 its-a-plane.py
**Traceback (most recent call last):
  File "/home/pi/its-a-plane-python/its-a-plane.py", line 1, in <module>
    from display import Display
  File "/home/pi/its-a-plane-python/display/__init__.py", line 7, in <module>
    from scenes.weather import WeatherScene
  File "/home/pi/its-a-plane-python/scenes/weather.py", line 7, in <module>
    from rgbmatrix import graphics
ModuleNotFoundError: No module named 'rgbmatrix'**
pi@raspberrypi:~/its-a-plane-python $

`
can someone give me a tip?

ColinWaddell, 2025-07-04:
It looks like it's not finding rgbmatrix. Have you been following this guide to get it installed and set up?

https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi/driving-matrices

cooool39, 2025-07-04:
i do and all the examples works great.

ColinWaddell, 2025-07-04:
So by the end of that installation process you should be able to run python from anywhere on your system and it can find rgb-matrix globally.

So if you run `python3` from the command line and then enter `from rgbmatrix import graphics` it should load the library. If that's not working I'd go through that installation guide again and check if anything's been missed.

It's been a while since I looked at it but I think that the demo applications you'll have ran are written in C, not python. So it's great the screen is being driven correctly, we just need to ensure that python also has access to the drivers that have been installed.

WiggyToo, 2025-07-07:
Hi Colin - thanks for all your hard work and design for the Fridge Flight Tracker - I've followed your excellent instructions and feel like I almost there (in my case I have combined it with this (https://www.raspberrypi.com/tutorials/build-your-own-raspberry-pi-flight-tracker/) so the Pi is doing both jobs.

I have the same problem as above, this is on a Pi 3 Model A+ - I have got as far as @cooool39 and have the same error.

```
plane@pi-radar:~/its-a-plane-python $ sudo python3 its-a-plane.py
Traceback (most recent call last):
  File "/home/plane/its-a-plane-python/its-a-plane.py", line 1, in <module>
    from display import Display
  File "/home/plane/its-a-plane-python/display/__init__.py", line 7, in <module>
    from scenes.weather import WeatherScene
  File "/home/plane/its-a-plane-python/scenes/weather.py", line 7, in <module>
    from rgbmatrix import graphics
ModuleNotFoundError: No module named 'rgbmatrix'
plane@pi-radar:~/its-a-plane-python $
```

I think there are some relevant points here, it's running Bookworm so it complains when you try to install the API

```
`sudo pip3 install FlightRadarAPI`

plane@pi-radar:~/its-a-plane-python $ sudo pip3 install FlightRadarAPI
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
plane@pi-radar:~/its-a-plane-python $
```

I went ahead and used --break-system-packages - I did tinker around trying to use VENV but as the hardware needs SUDO or root access my brain and knowledge failed me.

Trying your python test as above:

```
plane@pi-radar:~ $ python3
Python 3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from rgbmatrix import graphics
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'rgbmatrix'
>>>
```

Which I think confirms the issue with finding the module, I can check the expected location or do other tests as required, any help gratefully accepted.

Thanks,
Gav

cooool39, 2025-07-07:
Hi Colin and Gav,
i have found the Sollution for my Probleme.
i had to install rgbmatrix separately via "pip3 install rgbmatrix" and now its works.

And i had the same probleme with the FlightradarAPI, i have istall it with: "sudo pip3 install FlightRadarAPI --break-system-packages".
But i could not test it, because the wind has changed and all the Planes are avoids me ;)  .

ColinWaddell, 2025-07-07:
I think to avoid any confusion I should probably rebuild my own
flight-tracker and check the instructions still work against more modern OS
releases

If folk could confirm exactly which OS they’re sticking on their Pi then
I’ll try and match and see how I get on

cooool39, 2025-07-07:
> > Hi Colin and Gav, i have found the Sollution for my Probleme. i had to install rgbmatrix separately via "pip3 install rgbmatrix" and now its works.
> > And i had the same probleme with the FlightradarAPI, i have istall it with: "sudo pip3 install FlightRadarAPI --break-system-packages". But i could not test it, because the wind has changed and all the Planes are avoids me ;) .
> 
> Thanks for this - I tried similar but got this error:
> 
> ```
> Some possibly relevant errors from pip install:
>     ERROR: Could not find a version that satisfies the requirement rgbmatrix (from versions: none)
>     ERROR: No matching distribution found for rgbmatrix
> ```

Try this: 

[image]

WiggyToo, 2025-07-08:
Thanks Colin - that would be much appreciated, very happy to test the process this end if you want to message me any draft process.

I've got something displaying at least - just need to check it out and confirm I have the config correct.

[image]

Gav

ColinWaddell, 2025-07-08:
What you'll find if you don't run with sudo is the screen flickers a bit
more. It should still show flights, etc
--
Colin
colinwaddell.com <https://www.colinwaddell.com>

ColinWaddell, 2025-07-12:
I've got everything up and running at my end. I'll write up in the next couple of days but the short version is:

- Install rgb-matrix as per instructions
- Clone repo
- Make a virtualenv and activate it
- `pip install -r requirements.txt`
  - Where requirements.txt will look something like this: https://github.com/ColinWaddell/its-a-plane-python/blob/features/venv/requirements.txt
- Head into wherever you installed and build the rgb-matrix library to where the python bindings live (something like `/home/pi/rpi-rgb-led-matrix/bindings/python`)
- Run `pip install .`

This will let you run the venv version of python as root and it'll pick up all the libraries. Something like `sudo env/bin/python3 its-a-plane.py` 

There's a broken feature just now which I'll need to fix before I merge the updates into `master` -- when the device is looking for new data the default is to flash an LED on the matrix. What you can do alternatively though is blink a GPIO which is wired up to, in my case, an LED. With the latest set of libraries there seems to be some conflict regarding port ownership.

One workaround is to not run as root. This is what I'm experimenting with. To make the rgb-matrix driver happy and complain a little less you can run:

```sudo setcap 'cap_sys_nice=eip' /usr/bin/python3.11```

This grants python permission to set real-time scheduling priorities without needing to run as root.

I'm going to leave my box running like this for a while to see how it looks.

I'll report back once I've written up those instructions correctly.

WiggyToo, 2025-07-12:
Great, thanks, will check this out after you report back.

Gav

WiggyToo, 2025-07-13:
Just trying this now, only comments below, but all looks good to me:

At this step:

`pip install -r requirements.txt`

I get this warning (assume this relates to your GPIO point above).

`DEPRECATION: RPi.GPIO is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
`

For this line 

`sudo env/bin/python3 its-a-plane.py`

I used the following - was not sure which directory to be in, but this works from the home folder for the user, 'plane' in my case, 'fridge' is the name of my venv.

`sudo fridge/bin/python3 its-a-plane-python/its-a-plane.py`

For the service creation my venv location was slightly different but the following worked

```
[Unit]
Description=Its A Plane
After=network.target

[Service]
Type=simple
ExecStart=/home/plane/fridge/bin/python3 /home/plane/its-a-plane-python/its-a-plane.py
WorkingDirectory=/home/plane/its-a-plane-python
StandardOutput=append:/home/plane/plane.log
StandardError=append:/home/plane/plane.log
Restart=on-failure
User=plane
Environment=PYTHONUNBUFFERED=1

[Install]
WantedBy=multi-user.target
```

Thanks for the new process, much appreciuated.

Regards,
Gav

ColinWaddell, 2025-07-16:
Updated instructions are available here.

Does anyone fancy giving them a proof read, or even a shot on their system?

https://github.com/ColinWaddell/its-a-plane-python/tree/features/venv

Cheers

WiggyToo, 2025-07-17:
Will take a look, spotted this on the tracker today, we have the Royal International Air Tattoo quite close so expect all sorts over the weekend.

[image]

ryz1620, 2025-07-20:
I'm pleased to report that the instructions above worked, but I needed to manually create the asset folder and the requirements.txt. The things to resolve now are the display flickering, which isn't a massive issue, and the running on startup, which isn't working for some reason.

colin-benchmark, 2025-07-20:
Cool. Thanks for trying the instructions @ryz1620 

If you didn't have the assets folder and the requirements file it sounds like you've been working with the new instructions but you didn't pull in the code on that branch.

But, that problems solved itself as I've just merged that branch with `master` so you should just need to pull in the latest code. Probably worth trying to see if it helps with the flicker.

Doing the soldering-mod really helps with flicker, as does the step in the instructions which allows python to change the task priorities on the system. Another thing I've noticed is a huge swing in quality between screens, some of the cheaper ones are really sensitive to noise. Have a look at see if when you wiggle (or even touch) the ribbon cable wether you see a lot of flickering -- that's normally a sign that it's a cheaper screen. It'll save you a lot of time trying to debug things!

closed 2025-07-21 by ColinWaddell

WiggyToo, 2025-07-22:
Hi @ColinWaddell - I have not restested the final

> Updated instructions are available here.
> 
> Does anyone fancy giving them a proof read, or even a shot on their system?
> 
> https://github.com/ColinWaddell/its-a-plane-python/tree/features/venv
> 
> Cheers

Hi @ColinWaddell - I have not restested the updated instructions but from proof reading I have the following comments:

1. Bookwork should be Bookworm
2. There's a typo here "To avoid running as root and to grant python permission to set real-time scheduling priorities without requiring **elevatads** permissions, run the command"
3. That's it

Cheers,
Gav

ColinWaddell, 2025-07-22:
Awesome, thanks @WiggyToo I very much appreciate the help!

ColinWaddell, 2025-07-22:
The updated instructions have been merged into the `main` branch and I've also renamed the project from `its-a-plane` to `FlightTracker`