ColinWaddell/FlightTracker #21: Multiple aircraft not showing up properly.

yashmulgaonkar, 2025-08-11:
Hi @ColinWaddell,
Thanks for the excellent project!
We’re located near an airport with multiple parallel runways, and the current FlightTracker doesn’t display multiple flights clearly. Are there any plans to add support for this in a future update?

ColinWaddell, 2025-08-12:
The current method of working out what to display might not be the best if you're that close to the airport. I've got some tweaks you can try but it might send you over the FR24 API allowance.

Currently the code looks for all the planes within the boundary you set, then picks the 5 closest to you to display. Would you like to increase that 5 to display more?

ColinWaddell, 2025-08-13:
Giving this 24 hours before closing.