ColinWaddell/FlightTracker #22: Display stops updating due to grab_weather() blocking indefinitely

Philbru64, 2025-08-17:
A few days ago (I think it was 14 Aug) I noticed that the clock on my FlightTracker had stopped and in fact no other display elements were updating. Before this I had been running FlightTracker successfully for about 6 weeks. Rebooting the Raspberry Pi got things working again for a few minutes, but then problem came up again.

I traced the issue back to a function call on line 76 of /scenes/weather.py:
```
raw_data = urllib.request.urlopen(request).read()
```
It turns out that this request fails sometimes. The [python docs](https://stackless.readthedocs.io/en/v3.3.7-slp/library/urllib.request.html) speak of a global default timeout setting, but I now understand that this is not defined by default. A timeout can be specified as part of the function call and any errors can be caught. I checked roughly how long this function call nominally takes and it appears to be under 1s so I went for a 3s timeout to give some margin. When catching the errors, I got one of the following:
```
The read operation timed out
The handshake operation timed out
```

Accordingly, I edited grab_weather() in /scenes/weather.py (except I did not include error printing to align with rest of weather.py):
```
# Cache grabbing weather data
@lru_cache()
def grab_weather(location, ttl_hash=None):
    del ttl_hash  # to emphasize we don't use

    content = None

    try:
        request = urllib.request.Request(WEATHER_API_URL + location)
        raw_data = urllib.request.urlopen(request,timeout=3).read()
        content = json.loads(raw_data.decode("utf-8"))
        
    except:
        pass

    return content
```
This appears to work around the problem except that the temperature display goes blank for about a minute when an error crops up - I'll keep it like this for now so that I can monitor roughly how regularly this happens and in future I'll implement something to retry the request. 

Some info about my setup:
- Raspberry Pi Zero 2W
- Operating System: Raspbian GNU/Linux 11 (bullseye)
- Kernel: Linux 6.1.21-v7+
- Python 3.9.2

As mentioned above, FlightTracker was working happily for 6 weeks before this issue cropped up. The issue came up very soon after Virgin Media upped my internet connection speed so I initially wondered if it wasn't due to some connection instability, but I pinged taps-aff.co.uk repeatedly for an hour and didn't lose any packets (during that same hour I ran FlightTracker a couple of times with timeouts). I'm therefore not exactly sure why this issue suddenly started. Have any other users experienced this issue?

Philbru64, 2025-08-19:
Update:

- I did ping tests on the RPi for 5 hours today:
taps-aff.co.uk: 0.037% packet loss (8/21500)
google.com: 0.014% packet loss (3/21500)
For comparison I did ping tests on my PC during the same time period:
taps-aff.co.uk: 0.005% packet loss (1/21500)
google.com: 0.005% packet loss (1/21500)
It's probably safe to say that the RPi loses slightly more packets due to 2.4GHz wifi traffic.
- Over a 17 hour period I caught 87 errors, all thrown by urlopen():
81 of "The handshake operation timed out"
5 of "The read operation timed out"
1 of "Temporary failure in name resolution"
The errors are quite evenly distributed over the day.

ColinWaddell, 2025-08-20:
@Philbru64 I've added your update plus a little bit of retry logic to the code. I've got it running on my setup now and will leave it for a couple of days. If nothing weird happens then I'll get this merged

ColinWaddell, 2025-08-20:
Perfect timing. WeatherAPI.com is playing up which means testing this end-to-end wont work as my taps-aff server can't fulfil any requests.

Philbru64, 2025-08-21:
Haha! Classic! At least the app can handle those errors now! :)

Thanks for updating the code. The retry function is handy! I've just started using it on my end too and will report any issues.

Philbru64, 2025-08-27:
No problems on this end. It's been running happily since I uploaded the new code.